raspberrypi/pico-feedback #447: USB low level example - fails on RP2350 with extra endpoint added

matsobdev, 2025-01-29:
Slightly modified (2 `BULK` endpoints - `0x01` and `0x81`) - [dev_lowlevel.zip](https://github.com/user-attachments/files/18588232/dev_lowlevel.zip), https://github.com/raspberrypi/pico-examples/tree/master/usb/device/dev_lowlevel works fine running on RP2040 and 2350. Adding extra `BULK IN` at `0x82` - [dev_lowlevel2.zip](https://github.com/user-attachments/files/18588336/dev_lowlevel2.zip) and RP2350 fails to enumerate, but RP2040 works fine. Changing compiler from `ARM GCC 14.2.1`/`13.3.1`/`9.3.1` to `ARM LLVM 18.1.3` and RP2350 works (`19.1.5` is not). `RISCV GCC 14.2.0` works as well. Release, debug, doesn't matter. Since example and its modification seems to be fine, so I guess it is not an example or SDK issue, so landed here.

lurch, 2025-01-29:
ping @liamfraser

kilograham, 2025-01-29:
Moving to SDK if (and if i'm understanding correctly) whether it works or not is dependent on which compiler you use