raspberrypi/pico-feedback #450: Interupts Interfere With WS2812 PIO Implementation

joerossrobotics, 2025-02-18:
I'm using the WS2812 driver implementation from the `pico-examples` repo to control a strip of LEDs. Additionally, my setup includes an MCP2515 CAN controller for communication.

When polling the CAN controller in the main loop, the LED strip behaves as expected. However, when switching to an interrupt-driven approach using the MCP2515's interrupt, the LED strip begins to glitch. While the general animation pattern remains recognizable, it is noticeably distorted.

I suspect this issue may be due to interference between the interrupt routine and the PIO operation for driving the WS2812 LEDs.
Any insights or suggestions would be greatly appreciated!

joerossrobotics, 2025-02-20:
So the issue was that the interrupt handling interferes with the precise timing required for the ws2812 protocol. The solution was to disable interrupts while writing to the led strip, and then re-enable them. I'm not sure if this is worth addressing in the example or not?

lurch, 2025-02-20:
I guess you could try running the "writing to the led strip" on one core, and the "handling interrupts" on the other core?

I believe the examples in `pico-examples` are deliberately simple, in order to make them easy to understand. Obviously if they become more "comprehensive", it then becomes harder to see what it is that they're actually trying to demonstrate.

peterharperuk, 2025-07-09:
I can't see that there's anything for us to do here?

closed 2025-07-09 by peterharperuk