raspberrypi/pico-feedback #452: Linux RP2350 Builds Fail

jadczak, 2025-03-04:
The default example blink program fails to build on linux.  It fails on SDK 2.1 and 2.1.1

Trying to run cmake manually suggest that some unrecognized options are being passed to the compiler (`-m64` and `-mrelax-cmpxchg-loop`)

[image]

[image]

lurch, 2025-03-04:
> I'm not a CMAKE jockey so I don't know how to tell CMAKE to clear out the CFLAGS.

Me neither 😉  I asked one of my colleagues, and he pointed me at https://github.com/raspberrypi/picotool/blob/master/xip_ram_perms/CMakeLists.txt#L8-L14

jadczak, 2025-03-04:
```cmake
unset(ENV{CFLAGS})
unset(ENV{CXXFLAGS})
unset(ENV{LDFLAGS})
```

Works, but it is oddly temperamental (odd to me at least).  

From a clean project if I don't unset the flags and try to compile it fails (as expected) with the following:

```
Executing task: /home/alex/.pico-sdk/ninja/v1.12.1/ninja -C /home/alex/source/rp-pico/blink/build 

ninja: Entering directory `/home/alex/source/rp-pico/blink/build'
ninja: error: loading 'build.ninja': No such file or directory
```

Once I get the failure, if I unset the flags in my `CMakeLists.txt`, I'll get the same error.

Clearing out my build directory and removing the build directory altogether doesn't help (odd) _until I restart vscode_.  🤔 

At any rate, I have a path forward on my laptop as long as I can remember to unset my flags in my `CMakeLists.txt`.

Another workaround that occurred to me earlier today is to clear my flags in a terminal and launch vscode from there

i..e

```bash
$ CFLAGS="" CXXFLAGS=""
$ vscode
```

And then the terminals launched by vscode don't inherit the default flags from the system and the build works without adjusting the `CMakeLists.txt`.

Not sure what you want to do with this issue at this point.  I feel like `CMakeLists.txt` should probably have the flags unset by default since I would want my build to be insulated from the system.  

Considering that entire toolchain and sdk are controlled, being impacted by system environmental variables getting injected into the build seems undesirable (and has me questioning some of my build systems at work now, lol).

lurch, 2025-03-04:
> Once I get the failure, if I unset the flags in my CMakeLists.txt, I'll get the same error.

That's to be expected, as **lots** of files get written into the build folder; so whenever you edit files (e.g. `CMakeLists.txt`) that influence how `cmake` works, you often have to wipe out the build folder and create it afresh.

> Clearing out my build directory and removing the build directory altogether doesn't help (odd) until I restart vscode. 🤔

I dunno what might be happening there, but it sounds like vscode (or one of it's extensions) must be caching some information, which doesn't go away until you restart vscode? 🤷 

> At any rate, I have a path forward on my laptop

Good to hear 🥳 

BTW, I believe that
```console
$ CFLAGS="" CXXFLAGS=""
$ vscode
```
 can probably just be condensed to this?
```console
$ CFLAGS= CXXFLAGS= vscode
```

> Not sure what you want to do with this issue at this point.

I'll leave that up to @will-v-pi as https://github.com/raspberrypi/picotool and https://github.com/raspberrypi/pico-vscode are his babies 😊 

> Considering that entire toolchain and sdk are controlled, being impacted by system environmental variables getting injected into the build seems undesirable

In the 4 years since RP2040 was launched, I don't recall seeing any similar problem-reports to this one, so perhaps ClearLinux is just doing something a bit nonstandard compared to the majority of Linux distributions? 🐧

lurch, 2025-03-05:
Haha, I just noticed the "I use Python because real languages make my head hurt." in your profile - I assume that you're aware of https://www.raspberrypi.com/documentation/microcontrollers/micropython.html ? 🐍

jadczak, 2025-03-05:
I am aware, but I'm that special kind of stupid / stubborn that doesn't use anything but C for embedded coding, haha.

Well, that and while I've accepted that I'm a hack and a fraud, I'd hate to be a hypocrite too when I'm brow beating my coworkers "Knock it off with all these classes, templates, and overloading.  Just use C!" in the embedded codebases at work and then sneakily using micropython for my own silly side projects.  😅

will-v-pi, 2025-03-05:
> In the 4 years since RP2040 was launched, I don't recall seeing any similar problem-reports to this one, so perhaps ClearLinux is just doing something a bit nonstandard compared to the majority of Linux distributions? 🐧

Yeah, this seems to be a ClearLinux issue - I found an old thread on their forums about these flags causing issues for other compilations (https://community.clearlinux.org/t/the-default-value-of-cflags-cxxflags-may-cause-compile-error/104)

> ```bash
> $ CFLAGS= CXXFLAGS= vscode
> ```

This is probably the best solution - I don't think I'll add unsetting them to CMakeLists.txt by default, because some users will use valid flags to configure their builds this way. I'll change the title to reflect that this is a ClearLinux issue, and probably worth leaving this open if others have this issue.

lurch, 2025-03-05:
> I'll change the title to reflect that this is a ClearLinux issue, and probably worth leaving this open if others have this issue.

I'm going to slightly disagree with Will here, and close the issue. No point leaving it open if there's no further action to be taken from our side; and if other ClearLinux users run into this problem, they'll still be able to find and read the workaround here even if the issue is closed.

closed 2025-03-05 by lurch