raspberrypi/pico-feedback #455: Accessibility of documentation

corbane, 2025-03-31:
Hello, would it be possible to provide the information in an accessible format? HTML is preferred. Features like advanced search and translation for non-English speakers (or those like me who only understand one in five English words) are not available in PDF format.

For example, the "Getting Started with the Raspberry Pi Pico-Series" documentation is 56 pages long, and the "C/C++ SDK for Raspberry Pi Pico-Series" is 700 pages long. I don't think it would be useful to print all this documentation on an inkjet printer.
It would therefore be more appropriate to provide the information on the web in a more accessible and browser-friendly format.

lurch, 2025-03-31:
See also e.g. #161 and #292

Whilst it's not what you're directly asking for, the SDK API documentation **is** also available at https://www.raspberrypi.com/documentation/pico-sdk/

JamesH65, 2026-01-13:
I doubt we would release the source of the datasheets (We have moved to using Typst markdown for docs for those who are interested), but it's possible to compile typst into HTML (it's still under development by the typst people but mostly there), but how we would publish that I don't know.

PDF is a worldwide standard used by everyone, so it's no surprise we use it.