raspberrypi/pico-feedback #456: RP2350 Hardware Design Guide Resistor Labelling Mixup

madh4tter, 2025-03-31:
Between the Schematic and the PCB Layout snippets of the RP2350 Hardware Design Guide, R3 in Figure 3 Is mislabeled in Figure 4 (Labelled R5). 

[image]

[image]

Figure 15 (The RP2350A schematic), this resistor is labelled as R5, 

[image]

whereas in Figure 16 (RP2350B) it's labelled as R3.

[image]

lurch, 2025-03-31:
ping @dp111

katshann-raspberry, 2026-08-20:
These diagrams have been updated and the fix will be included in the next published version of the PDF.

closed 2026-08-20 by katshann-raspberry