raspberrypi/pico-feedback #459: https://www.usbmadesimple.co.uk/ is down...

matsobdev, 2025-05-02:
...for sale at this moment. I guess it should be excluded from datasheets recomendations on how to get familiar with USB.

lurch, 2025-05-06:
😢 
Searching Google for usbmadesimple still shows cached results from the previous version of the site, so _hopefully_ this is just a temporary glitch, and the old site will be restored at some point? 🤞

lurch, 2025-05-19:
> The old content is still available at https://web.archive.org/web/20250415152420/https://www.usbmadesimple.co.uk/

...which shows that the usbmadesimple website was owned by a company called "MQP Electronics Ltd", which apparently had their website at www.mqp.com but that too is also now down. And searching for the company name also finds https://find-and-update.company-information.service.gov.uk/company/02563219 which says that the company was dissolved on
31 March 2025, which obviously explains why both of those websites are down, and is a good indicator that they'll probably never be coming back 😢  I'll discuss this with my colleagues tomorrow...

lurch, 2025-05-21:
Thank you for the report. This has been fixed in our (private) sourcecode, and will be visible whenever we next publish updated versions of the RP2040 and RP2350 datasheets.
After a bit of discussion, we decided to go with @jannic 's suggestion of just linking to the `archive.org` version of the site.