raspberrypi/pico-feedback #461: Questtion: Raspberry Pi Pico-series C/C++ SDK Dokumentation

hasenradball, 2025-05-06:
In the chapter 3.2 Getting Started with pio on page 46 last lines.

There is the following function described:

```
sm_config_set_out_shift(&c, false, true, rgbw ? 32 : 24);
False for shift_to_right (i.e. we want to shift out MSB first).
True for autopull. 32 or 24 for the number of bits for the autopull threshold, i.e. the point at which the state
machine triggers a refill of the OSR, depending on whether the LEDs are RGB or RGBW
```

Is it a typo???

If we want to shift MSB first we need to shift to left, right?

lurch, 2025-05-06:
I believe that it's saying "we pass `false` to the `shift_to_right` parameter" (i.e. it's shifting to the left, which agrees with what you're suggesting?)

hasenradball, 2025-05-06:
Dear lurch,

just writing in short what I think.
Somtimes it is very hard to understand what is meant directly, because here I don't see the parameter names of the function, so don't know that the parameter is called `shift_to_right`.
But would try to describe it more clearly.

```
sm_config_set_out_shift(&c, false, true, rgbw ? 32 : 24);
   &c : adress of pio config struct
   false : leads to shift to the left (we want to shift MSB first here), otherwise true for shift to the right
   true : set autopull to true
   32 or 24 : for the number of bits for the autopull threshold, i.e. the point at which the state
machine triggers a refill of the OSR, depending on whether the LEDs are RGB or RGBW
```
Small change but from my point of view a bit clearer.

lurch, 2025-05-06:
> I don't see the parameter names of the function, so don't know that the parameter is called `shift_to_right`

See e.g. https://www.raspberrypi.com/documentation/pico-sdk/hardware.html#group_sm_config_1ga613bed03e10e569f1b7aede74d40a5b5  (and yes, I've already put in a (unpublished) fix to update `shift_to_right` to `shift_right` 😇 )

hasenradball, 2025-05-06:
Sounds good.

Is there a Link to the function description in the Pico documentation?

lurch, 2025-05-07:
See page 237 of https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf

closed 2025-07-09 by peterharperuk