raspberrypi/pico-feedback #466: RP2350 datasheet: Parameters for REBOOT_TYPE_BOOTSEL are swapped

iwanders, 2025-07-13:
In the RP2350 datasheet[^1], section `5.4.8.24. reboot`, the `REBOOT_TYPE_BOOTSEL` parameters are described as:
-  Signature: ` int reboot(uint32_t flags, uint32_t delay_ms, uint32_t p0, uint32_t p1)`
- `p0`: - the GPIO number to use as an activity indicator
- `p1`: a set of flags (most notably, `0x01` for disabling the mass storage device).

I believe this is the incorrect order and it should read that the first argument `p0` holds the flags, and the second argument `p1` holds the gpio number.

The pico-sdk calls the function correctly: in [rom_reset_usb_boot](https://github.com/raspberrypi/pico-sdk/blob/9a4113fbbae65ee82d8cd6537963bc3d3b14bcca/src/rp2_common/pico_bootrom/bootrom.c#L43-L44), where it uses `rom_reboot(TYPE_BOOTSEL, delay, flags, gpio);`, indicating that `p0` should be flags and `p1` the gpio pin (similarly on line 63). The `rom_reboot` function [wrapper](https://github.com/raspberrypi/pico-sdk/blob/9a4113fbbae65ee82d8cd6537963bc3d3b14bcca/src/rp2_common/pico_bootrom/include/pico/bootrom.h#L502-L505) does not perform any argument swaps. The [doxygen](https://github.com/raspberrypi/pico-sdk/blob/9a4113fbbae65ee82d8cd6537963bc3d3b14bcca/src/rp2_common/pico_bootrom/include/pico/bootrom.h#L469-L475) there incorrectly follows the datasheet, the implementation has the correct order.

Note that the similar rp2040 bootrom method `_reset_to_usb_boot` (rp2040-datasheet, p135) does use `gpio_activity_pin_mask` as first argument, and `disable_interface_mask` as second.

I discovered this after trying to disable the mass storage device when entering BOOTSEL mode through this bootrom method, I couldn't get it to work. After reading `The p0 and p1 parameters are generally written to watchdog scratch registers 2 & 3` at the bottom of the section (p396) I tried setting it up through the watchdog, I wasn't sure if this 'human readable' sentence meant the 2nd and 3rd scratch register (so `SCRATCH1` and `SCRATCH2`) or the zero indexed ones, so `SCRATCH2` & `SCRATCH3`. I tried both, and putting `DISABLE_MSD_INTERFACE` into `SCRATCH2` worked, with that information I tried swapping the order on the reboot bootrom method, and that also worked. Only after that did I trace through the pico-sdk to see how its done there, which provides confidence the datasheet has the order wrong. I've confirmed this behaviour on two different Pico 2W boards, both revision A2, and ruled out problems with the Rust HALs by testing this problem with the pico-sdk using [this](https://gist.github.com/iwanders/3964bd9ca7002d45dd0b17dd76ec64ef) simple program.

[^1]: Datasheet retrieved from [the official site](https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf), `build-date: 2025-02-20`, `build-version: 3184e62-clean`, retrieved today.

lurch, 2025-07-13:
The [`s_varm_api_reboot`](https://github.com/raspberrypi/pico-bootrom-rp2350/blob/master/src/main/native/bootrom.h#L109) function in the [RP2350 bootrom sourcecode](https://github.com/raspberrypi/pico-bootrom-rp2350) also has the signature `int s_varm_api_reboot(uint32_t flags, uint32_t delay_ms, uint32_t p0, uint32_t p1)`, and https://github.com/raspberrypi/pico-bootrom-rp2350/blob/master/src/main/arm/varm_boot_path.c#L894 _maybe_ suggests that for `REBOOT2_FLAG_REBOOT_TYPE_BOOTSEL`, the `bootsel_flags` parameter comes before the `gpio_pin_config` parameter? 🤔 
(please note that I've never looked at the bootrom code before, so this is just a bit of a wild guess! 😉 )

kilograham, 2025-07-17:
this is fixed and will appear in the next docs release

closed 2025-07-17 by kilograham